bitluni/ESP32CompositeVideo #2: Flashing image on car / FPV screen

AJRussell, 2018-03-09:
Hi Bitluni,

I have posted this on your Patron as well, I hope you don't mind me posting here too.

I have tried the Composite video on my -ordinary TV- and it works perfectly but when I try it on a car monitor device like this one: https://www.amazon.co.uk/gp/product/B0056WJP1M it shows the image but flashes on and off about twice a second, would you have any idea why this would be?

I have also tried a FPV screen and had similar results.

Thanks for any help or advice.

Anthony.

Cyb3rn0id, 2018-09-09:
I've the same issue but on all my TV. I've tried on a new LCD, on an old CRT and on an oldest CRT from Apple (green phospor used for an old Apple Computer). 
I see only an horizontal line flashing on All TVs. My TVs are all PAL, so I've changed the line:

CompositeOutput composite(CompositeOutput::PAL, XRES * 2, YRES * 2);

and the:
const int XRES = 320;
const int YRES = 240;

I'm using a Wemos Lolin32

JyeSmith, 2018-11-16:
@AJRussell  and @Cyb3rn0id did either of you find a fix?  It works for me on a CRT TV and FPV goggles, but not on a small 4.3" TFT screen I use for testing.

AJRussell, 2018-11-16:
@JyeSmith  Hi, no I could not get the Tetris Space x code working but the Dawn of AV code worked fine.
I believe bitluni will be revising the code and creating a Arduino library, so hopefully it will work then.
Its interesting that it works for you on FPV goggles..

JyeSmith, 2018-11-17:
@AJRussell it isn't perfect with goggles.  Skyzones have some burring/ghosting, and it is worse on Fatsharks.

I look forward to the lib!