bitluni/ESP32CompositeVideo #3: ESP32-D2WD 2 megabytes flash RAM & resolution

bootrino, 2018-03-13:
Hello,

I read that the ESP32-D2WD has 2 megabytes of embedded flash memory.

Would this be usable as graphics memory for your ESP32CompositeVideo project or would it be too slow?

bitluni, 2018-03-13:
you mean PSRAM not flash (as on the Wrover modules)
it's certainly not usable as video memory since the DMA for I2S is limited to a specific address range.
the speed of the spi is also limited making it hard to copy parts of frames into DMA ram in time.
some cache warming could make it possible to hold parts of the frame buffer in the PSRAM. But this is quite complicated and really hardware dependent.. newer revisions of the chip could break it.
However the PSRAM comes handy for all other stuff like meshes, vertices to free up some DAM ram

bootrino, 2018-03-13:
Hey there bitluni great great work love your videos.

>> it's certainly not usable as video memory since the DMA for I2S is limited to a specific address range.
Oh that's a pity not to have enough RAM for larger color images.

>> you mean PSRAM not flash
I'm pretty sure it's flash on the chip for the D2WD model.
This datasheet: https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf

says:
"ESP32-D2WD has a 16-Mbit, 40-MHz embedded flash, connected via pins GPIO16, GPIO17, SD_CMD, SD_CLK, SD_DATA_0 and SD_DATA_1."

closed 2018-03-26 by bootrino