bitluni/ESP32CompositeVideo #4: Not working with WiFi.h

dsiberia9s, 2018-03-15:


dsiberia9s, 2018-03-15:
server.begin();
TVout::begin();

work only ONE or server or tvout

dsiberia9s, 2018-03-15:
namespace TVout
{
  #include <soc/rtc.h>
  #include "utility/CompositeGraphics.h"
  #include "utility/Image.h"
  #include "utility/CompositeOutput.h"
  #include "utility/font6x8.h"
  
  const int XRES = 320;
  const int YRES = 200;
  
  CompositeGraphics graphics(XRES, YRES);
  CompositeOutput composite(CompositeOutput::NTSC, XRES * 2, YRES * 2);
  Font<CompositeGraphics> font(6, 8, font6x8::pixels);

  void compositeCore(void *data) {
    while (true)
    {
      composite.sendFrameHalfResolution(&graphics.frame);
    }
  }
  
  void begin() {
    rtc_clk_cpu_freq_set(RTC_CPU_FREQ_240M);
    composite.init();
    graphics.init();
    graphics.setFont(font);
    xTaskCreatePinnedToCore(compositeCore, "c", 1024, NULL, 1, NULL, 0);
  }

  void setTextColor(int c) {
    graphics.setTextColor(c);
  }
  
  void setCursor(int x, int y) {
    graphics.setCursor(x, y);
  }

  void print(char *str) {
    graphics.print(str);
  }

  void drawBitmap(int x, int y, int w, int h, const unsigned char* img) {   
    Image<CompositeGraphics> img_(w, h, img);
    img_.draw(graphics, x, y);
  }

  void fillRect(int x, int y, int w, int h, int color = 0) {
    graphics.fillRect(x, y, w, h, color);
  }
}

PerroLoco64, 2019-12-15:
I think that the problem it´s because the CompositeVideo needs to constantly receive the signal to see a proper image, that´s why bitlunis put that proces in core 0 (normal code runs in core 1). The fact its that Wifi process also runs in core 0, so when you add Wifi to your project, now there are some delays  when the ESP32 sends the signal. A solution will to put the wifi procces in core 1 but I ddin't find how to do it in Arduino IDE only on ESP-IDF.

Sorry if my english is bad

wgaylord, 2020-08-04:
You could put the composite.sendFrameHalfResolution(&graphics.frame); inside of loop() and do your actual drawing and what not on core 0. So that the only thing on core 1 is the composite.sendFrameHaldResolution in the loop.

closed 2023-06-17 by dsiberia9s