bitluni/ESP32CompositeVideo #8: Arduino 1.0 stable. Crashes

IlyaMZP, 2018-08-21:
```ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
***ERROR*** A stack overflow in task c has been detected.
abort() was called at PC 0x400882a4 on core 0

Backtrace: 0x40088088:0x3ffdc9d0 0x4008828b:0x3ffdc9f0 0x400882a4:0x3ffdca10 0x4008529b:0x3ffdca30 0x400869c0:0x3ffdca50 0x40086976:0xfefefefe

Rebooting...
```

IlyaMZP, 2018-08-21:
```E (203) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/i2s.c:1141 (i2s_write):tx NULL
E (204) I2S: /Users/ficeto/D***ERROR*** A stack overflow in task c has been detected.
```
Latest github snapshot

marcel303, 2018-10-29:
Same issue as https://github.com/bitluni/ESP32CompositeVideo/issues/10
@IlyaMZP Could you please try the suggested fix in https://github.com/bitluni/ESP32CompositeVideo/issues/10 and report back? Thanks.

marcel303, 2018-10-31:
@IlyaMZP The fix above has been merged, so it should work just syncing to the latest version!

closed 2018-11-07 by IlyaMZP