bitluni/ESP32CompositeVideo #10: ***ERROR*** A stack overflow in task c has been detected.

NuclearManD, 2018-10-08:
Hello, just uploaded buth sketches (unmodified) from this repo onto my Sparkfun ESP32 dev board.  It keeps rebooting, here is the serial monitor output:

`ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
***ERROR*** A stack overflow in task c has been detected.
abort() was called at PC 0x4008a2e4 on core 0

Backtrace: 0x4008a0c8:0x3ffdad00 0x4008a2cb:0x3ffdad20 0x4008a2e4:0x3ffdad40 0x400872db:0x3ffdad60 0x40088a00:0x3ffdad80 0x400889b6:0xfefefefe

Rebooting...
(repeats)`

I just got this program to play with it, but it doesn't work the first time.

Arduino version is 1.8.5.   Not sure about ESP32 core version, but I downloaded that into Arduino/hardware manually.  If I remember correctly, I downloaded it from the ESP32 Arduino core github.

You probably don't need this, but here are my upload settings:

Flash Mode:  QIO
Flash Size: 4MB
Partition Scheme: Minimal (tried also with Default, same problem)
Flash Frequency: 80Mhz
PSRAM: Disabled
Upload Speed: 921600
Core Debug Level: None

I can't imagine there would be any recursive behavior or large stack allocations to cause this, so it seems pretty odd.

Unfortunately I just downloaded this and I have zero understanding of the code thus far.

Here is the zip I downloaded.  The only modification I made was to the sketch name, just to make it more unique. I do not think that would be the problem though.
[ESP32CompositeVideo-master.zip](https://github.com/bitluni/ESP32CompositeVideo/files/2456206/ESP32CompositeVideo-master.zip)

marcel303, 2018-10-29:
@NuclearManD I ran into the same issue but was able to resolve it by changing this line,
    xTaskCreatePinnedToCore(compositeCore, "c", 1024, NULL, 1, NULL, 0);

To this,
   xTaskCreatePinnedToCore(compositeCore, "c", 2048, NULL, 1, NULL, 0);

The third parameter of xTaskCreatePinnedToCore is the stack size, which is like a scratch space created for each task. Apparently using a newer version of the ESP32 core uses more scratch itself? Or something else has changed? In either case, it needs to be bumped to 2048 to work.

Can you check if this works for you and report back?

It works for me, but after that I get another error; this one more hard to fix. I get this message repeatedly on the serial monitor,
"/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/i2s.c:1141 (i2s_write):tx NULL"

Which seems to indicate the dma queue for the DAC hasn't been set up properly. But I see zero errors during i2s/DAC initialization..

marcel303, 2018-10-29:
Oddly, ESP32SpaceShooter does work. I haven't found any significant changes to how it sets up the DAC yet! @bitluni are ESP32SpaceShooter and ESP32CompositeVideo more or less equivalent when it comes to composite output, or can you maybe think of something which could explain why the space shooter code works fine but the code for ESP32CompositeVideo giving the "tx NULL" issue above?

marcel303, 2018-10-31:
@NuclearManD @bitluni I managed to fix the second issue ("tx NULL") as well. Without going into too much detail, in sendFrameHalfResolution it sends a bunch of lines. The code uses a variable 'i' to write into the array of pixels for a single line. 'i' gets reset at the start of each line, but there is one instance where (I think) it is not, causing the fill routines to overflow the line array, and causing Bad Things To Happen elsewhere. The "tx NULL" is just a side effect of this. I will make a pull request soon.

marcel303, 2018-10-31:
The stack overflow is also magically fixed when fixing the buffer overflow.. :-)

marcel303, 2018-10-31:
Pull request: https://github.com/bitluni/ESP32CompositeVideo/pull/11

closed 2018-10-31 by bitluni

NuclearManD, 2018-11-01:
Erm, this did not fix it for me :/  My tv still gets no output and my serial monitor is spammed with these messages:

E (3641) I2S: /Users/ [redacted for someones privacy] /Desktop/ESP32/ESP32/esp-idf-public/components/driver/i2s.c:1141 (i2s_write):tx NULL

marcel303, 2018-11-01:
Did you try the CompositeVideoSimple sketch or CompositeVideo? I just noticed the CompositeVideoSimple sketch and made a fix for the second sketch as well. CompositeVideo should work; try it.

NuclearManD, 2018-11-09:
Sorry for such late reply; I am very busy these days.  I tested both, neither work.  Perhaps a new commit has a fix?  I cannot test at the moment, will try to later.

NuclearManD, 2018-11-09:
Just retested, still same problem.  No signal from my TV and the serial monitor is flooded.  Good to hear that you are revisiting it, it is an awesome project! :)

JyeSmith, 2018-11-16:
> 
> 
> I'm revisiting the whole project in the next weeks

This isnt of use for your current projetc, but have you (or anyone else) looked at how to use this lib as an OSD?  With or without the use of an lm1881?

Cheers