bitluni/ESP32CompositeVideo #13: Is it possible to capture composite video with esp32 (and send picture via WiFi)

X-NoNAME, 2019-01-31:
I have an old camera with composite video out. And I think to start use it.

Sorry for  my English. :smile:

desegura, 2020-10-07:
Interested too please @bitluni some advice??? 😋

wgaylord, 2021-02-04:
You may be able to use the I2S with the ADC to capture the frame voltages and sort of reverse it. No idea how it would work but its an idea.

entozoon, 2021-11-11:
This isn't really relevant to this project; it's basically the opposite concept.
That said, it has been discussed elsewhere e.g. 
https://forum.arduino.cc/t/capturing-a-still-image-from-a-composite-video-signal/377162/8
https://arduino.stackexchange.com/questions/66271/is-it-possible-to-convert-rcavideo-only-to-arduino-uno-then-to-a-computer