bitluni/ESP32CompositeVideo #14: Ideas for how to synchronize to an external video source?

AlessandroAU, 2019-03-20:
Hi,

I want to be able to synchronize this to an external video signal so I can generate an OSD overlay.

But the synchronization part is seemingly impossible!

I'm made a test circuit with an LM1881 sync separator and I have modified the composite output library so that the clock source for the I2S DMA is set to the APLL. I can then modify the APLL registers with rtc_clk_apll_enable() and have very fine frequency control over the I2C DMA output. 

I then wrote a Phase Frequency Detector type PLL 'loop' in order to lock onto the external VSYNC signal. It compares the VSYNC pulse to when data is written I2C DMA and tries to get them into sync by adjusting the the APLL regs and hence the frequency that the data is pushed out. 

I can get reasonably close to a 'lock' but no matter what I do there is always about 1 line of jitters which is not acceptable for drawing an OSD. 

I also tried a 'dumb' solution where the external VSCYN simply triggers the function to write to the I2C buffer but this still jitters badly....

Anyone with an ideas, please help, I am stuck :(

JyeSmith, 2019-03-20:
This would be an awesome feature to have in this library!

Surubel, 2019-05-09:
Maybe the problem is in hardware. Would you care to share the setup? I agree that this would be awesome, and I intend to make a development board centered around composite video on the esp32. Any prior experience will be of help.

RD000000, 2022-01-12:
That may be instability in PLL loop, common problem for any feedback systems. Try to lower gain over the loop. Or, better, add PID regulator to try proportional, integral, differential gains.

AlessandroAU, 2022-01-12:
PR please :)

RD000000, 2022-01-12:
Right now looking at code :)

Surubel, 2022-01-12:
Almost forgot about this, now that I've re-read the issue, what kind of source did you use? Is there any possibility that it might be interlaced? This could be causing exactly 1 line of jitter, but it would have to be constant. 
Haven't looked properly at the code yet but in the meantime I got a few prototypes of the devboard with ESP32 and LM1881, but never got around to writing the code for any such functionality. Maybe if I get around to it this week I could try running AlessandroAU's code on it, if that would be ok.

AlessandroAU, 2022-01-12:
I don't think it was an interlacing problem, the jitter was much more than a single line and it was also side-to-side not just vertical. I'll publish my code to get you going.

DiscoMan18, 2023-01-07:
Alessandro, have you finally achieved any progress with that jitter issue? I was thinking on similar project (LM1881, ESP32 and composite video) but then found your code

DiscoMan18, 2023-01-08:
Also, what revision of chip were you working with? If it was 1.0 then according to this documents APLL has bugs in calculating correct frequency as mentioned [here](https://www.espressif.com/sites/default/files/documentation/eco_and_workarounds_for_bugs_in_esp32_en.pdf)

Also found this link on the matter: https://www.esp32.com/viewtopic.php?t=3176 Take a look at it. Hope this helps. Regarding you project, could you please provide a schematic of it?