bitluni/ESP32CompositeVideo #16: rtc.h not found.

gabrielcrowe, 2019-06-09:
I have installed the esp32 requirements and i get an error on compile for this:

```
AudioVideoExample:3:21: error: soc\rtc.h: No such file or directory
 #include <soc\rtc.h>
                     ^
compilation terminated.
exit status 1
soc\rtc.h: No such file or directory
```

An anybody help?

gabrielcrowe, 2019-06-09:
I discovered a catch 22 in this problem:

This file is **not** included if I use the esp8266 libraries

It is included if i use the esp32 ones.

But the esp32 ones use a different version of esptool and this version cannot connect to my ai-thinker esp8266.

in general, i cannot get this to work with the esp8266 dev boards you can buy on amazon these days because of this problem.

Fredyy90, 2020-06-01:
This library is made for the ESP32, so its correct that it wont work on esp8266 out of the box.
It might even be impossible to port the code, since the esp8266 has a lot less computational power.

bitluni, 2020-06-01:
it's esp32 only. esp8266 doesn't have a dac

closed 2026-08-19 by leenattress