bitluni/ESP32CompositeVideo #23: Arduino 1.8.13 and 1.8.9 not compiling

BradTotaro, 2021-03-08:
Arduino: 1.8.9 (Linux), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

In file included from /home/brad/ESP32CompositeVideo/CompositeVideo/CompositeVideo.ino:9:0:
/tmp/arduino_build_296062/sketch/Matrix.h:28:31: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
   const float operator[](int i)
                               ^
/home/brad/ESP32CompositeVideo/CompositeVideo/CompositeVideo.ino: In function 'void draw()':
/home/brad/ESP32CompositeVideo/CompositeVideo/CompositeVideo.ino:148:33: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   graphics.print("free memory: ");
                                 ^
/home/brad/ESP32CompositeVideo/CompositeVideo/CompositeVideo.ino:150:26: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   graphics.print(" fps: ");
                          ^
/home/brad/ESP32CompositeVideo/CompositeVideo/CompositeVideo.ino:152:34: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   graphics.print(" triangles/s: ");
                                  ^
In file included from /home/brad/ESP32CompositeVideo/CompositeVideo/CompositeVideo.ino:11:0:
/tmp/arduino_build_296062/sketch/Mesh.h: In instantiation of 'Mesh<Graphics>::Mesh(int, const float (*)[3], int, const short unsigned int (*)[2], int, const short unsigned int (*)[3], const float (*)[3]) [with Graphics = CompositeGraphics]':
/home/brad/ESP32CompositeVideo/CompositeVideo/CompositeVideo.ino:40:141:   required from here
Mesh.h:11:28: error: 'Mesh<CompositeGraphics>::vertices' will be initialized after [-Werror=reorder]
   const float (*vertices)[3];
                            ^
Mesh.h:10:7: error: 'int Mesh<CompositeGraphics>::edgeCount' [-Werror=reorder]
   int edgeCount;
       ^
Mesh.h:18:3: error: when initialized here [-Werror=reorder]
   Mesh(int vertCount, const float verts[][3], int edgeCount_ = 0, const unsigned short edges_[][2] = 0, int triCount = 0, const unsigned short tris[][3] = 0, const float triNorms[][3] = 0)
   ^
Mesh.h:16:34: error: 'Mesh<CompositeGraphics>::edges' will be initialized after [-Werror=reorder]
   const unsigned short (*edges)[2];
                                  ^
Mesh.h:9:7: error: 'int Mesh<CompositeGraphics>::triangleCount' [-Werror=reorder]
   int triangleCount;
       ^
Mesh.h:18:3: error: when initialized here [-Werror=reorder]
   Mesh(int vertCount, const float verts[][3], int edgeCount_ = 0, const unsigned short edges_[][2] = 0, int triCount = 0, const unsigned short tris[][3] = 0, const float triNorms[][3] = 0)
   ^
Mesh.h:15:38: error: 'Mesh<CompositeGraphics>::triangles' will be initialized after [-Werror=reorder]
   const unsigned short (*triangles)[3];
                                      ^
Mesh.h:12:35: error: 'const float (* Mesh<CompositeGraphics>::triangleNormals)[3]' [-Werror=reorder]
   const float (*triangleNormals)[3];
                                   ^
Mesh.h:18:3: error: when initialized here [-Werror=reorder]
   Mesh(int vertCount, const float verts[][3], int edgeCount_ = 0, const unsigned short edges_[][2] = 0, int triCount = 0, const unsigned short tris[][3] = 0, const float triNorms[][3] = 0)
   ^
cc1plus: some warnings being treated as errors
exit status 1
'Mesh<CompositeGraphics>::vertices' will be initialized after [-Werror=reorder]

vslinuxdotnet, 2021-06-01:
Some problem there +1

vslinuxdotnet, 2021-06-01:
The problem is the order of variables in the instance Mesh class.
change the order in top of the Mesh.h file to:

```
template<class Graphics>
class Mesh
{
  public:
  int vertexCount;
  const float (*vertices)[3];
  int edgeCount;
  const unsigned short (*edges)[2];
  int triangleCount;
  const unsigned short (*triangles)[3];
  const float (*triangleNormals)[3];
  short (*tvertices)[3];
  signed char (*tTriNormals)[3];
```

Regards
VS