bitluni/ESP32CompositeVideo #32: Not working with ESP32 Arduino Core >2.0.0 stable

thijses, 2022-09-07:
I noticed that the compiler errors/warnings exist in version 2.0.0 of the esp32 core for Arduino IDE, but in versions 2.0.4, 2.0.3 (2.0.2 and 2.0.1 not tested), it just doesn't display anything. 2.0.0 still works though.

thijses, 2022-09-07:
(to be clear, by "work" i mean actually output a signal. I think Espressif changed the API for I2S (if the compiler warnings are to be believed) None of the versions spit out an error, and turning on debug outputs (from None to something like Verbose), sometimes removes the compiler warnings (which is bad, because why would that solve anything???))

Erhannis, 2022-11-26:
Thanks; I was using 2.0.5 (I think) and it wasn't working.  I downgraded to 1.0.6 (because I misread your post) and it worked.  My oscilloscope says that with the new version, the signal varies between only like 90mv and 120mv or something small like that, but with the old version it varies, uh, I think somewhat lower, but all the way up to like 1.2v (about as expected).  Not sure if that just means the new version is outputting blank data, or if it's scaled wrong, or what.

whzlxa, 2023-03-19:
still not work when i was using 2.0+ IDE and 1.8.X IDE, PIN 25 with noting output seems  somting wrong?

BobSleigh38, 2024-01-26:
Hello
I'm also having issues with composite video
I can't see anything starting from espressif v2.0.1
v2.0.0 is ok ....