bitluni/ESP32CompositeVideo #35: Not working on my esp32-wroom-32u board connected to crt

AyceBenz, 2024-01-13:


BobSleigh38, 2024-01-26:
Hello
I'm also having issue with composite video
What worked for me is to downgrade espressif boards in Arduino board manager IDE to an older version.
I use 2.0.0 instead of 2.0.14
Something happened at v. 2.0.1 ....

n-92, 2024-06-15:
Meanwhile, I am getting this bunch of error on my serial monitor. 
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d23ec  PS      : 0x00060b30  A0      : 0x800d2f1a  A1      : 0x3ffb2780  
A2      : 0x3ffbffd4  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x000002c0  
A6      : 0x00000000  A7      : 0x00000218  A8      : 0x000000b0  A9      : 0x00000001  
A10     : 0x3ffb6200  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x3ffc076c  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000017  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x40085874  LEND    : 0x4008588a  LCOUNT  : 0xffffffff  

Backtrace:0x400d23e9:0x3ffb27800x400d2f17:0x3ffb2800 0x400d33c5:0x3ffb2820 

ELF file SHA256: 0000000000000000