bitluni/ESP32CompositeVideo #37: 'I2S_CLKM_CONF_REG' was not declared in this scope

Giak88, 2024-08-04:
I've installed all necessary library, also the i2s.h file from the 'esp32'.

In the CompositeOutput.h cannot compile due to 'I2S_CLKM_CONF_REG' was not declared in this scope.

I've not touched the code since I want it running as a test.

Here's the full error:
`Arduino:1.8.19 (Windows 10), Scheda:"ESP32 Dev Module, Disabled, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled, Disabled"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\giaco\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\giaco\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:JTAGAdapter=default,PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=none,EraseFlash=none,ZigbeeMode=default -ide-version=10819 -build-path C:\Users\giaco\AppData\Local\Temp\arduino_build_407299 -warnings=none -build-cache C:\Users\giaco\AppData\Local\Temp\arduino_cache_457876 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esp-x32.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\esp-x32\2302 -prefs=runtime.tools.esp-x32-2302.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\esp-x32\2302 -prefs=runtime.tools.dfu-util.path=C:\Users\giaco\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5 -prefs=runtime.tools.dfu-util-0.11.0-arduino5.path=C:\Users\giaco\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5 -prefs=runtime.tools.esp-xs3.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\esp-xs3\2302 -prefs=runtime.tools.esp-xs3-2302.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\esp-xs3\2302 -prefs=runtime.tools.openocd-esp32.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.12.0-esp32-20240318 -prefs=runtime.tools.openocd-esp32-v0.12.0-esp32-20240318.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.12.0-esp32-20240318 -prefs=runtime.tools.esptool_py.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.6 -prefs=runtime.tools.esptool_py-4.6.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.6 -prefs=runtime.tools.mkspiffs.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esp-rv32.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\esp-rv32\2302 -prefs=runtime.tools.esp-rv32-2302.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\esp-rv32\2302 -prefs=runtime.tools.esp-xs2.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\esp-xs2\2302 -prefs=runtime.tools.esp-xs2-2302.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\esp-xs2\2302 -prefs=runtime.tools.mklittlefs.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.xtensa-esp-elf-gdb.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\12.1_20231023 -prefs=runtime.tools.xtensa-esp-elf-gdb-12.1_20231023.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\12.1_20231023 -prefs=runtime.tools.riscv32-esp-elf-gdb.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gdb\12.1_20231023 -prefs=runtime.tools.riscv32-esp-elf-gdb-12.1_20231023.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gdb\12.1_20231023 -prefs=runtime.tools.esp32-arduino-libs.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67 -prefs=runtime.tools.esp32-arduino-libs-idf-release_v5.1-dc859c1e67.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67 -verbose D:\Downloads\ESP32CompositeVideo-master\ESP32CompositeVideo-master\CompositeVideoSimple\CompositeVideoSimple.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\giaco\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\giaco\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:JTAGAdapter=default,PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=none,EraseFlash=none,ZigbeeMode=default -ide-version=10819 -build-path C:\Users\giaco\AppData\Local\Temp\arduino_build_407299 -warnings=none -build-cache C:\Users\giaco\AppData\Local\Temp\arduino_cache_457876 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esp-x32.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\esp-x32\2302 -prefs=runtime.tools.esp-x32-2302.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\esp-x32\2302 -prefs=runtime.tools.dfu-util.path=C:\Users\giaco\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5 -prefs=runtime.tools.dfu-util-0.11.0-arduino5.path=C:\Users\giaco\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5 -prefs=runtime.tools.esp-xs3.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\esp-xs3\2302 -prefs=runtime.tools.esp-xs3-2302.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\esp-xs3\2302 -prefs=runtime.tools.openocd-esp32.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.12.0-esp32-20240318 -prefs=runtime.tools.openocd-esp32-v0.12.0-esp32-20240318.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.12.0-esp32-20240318 -prefs=runtime.tools.esptool_py.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.6 -prefs=runtime.tools.esptool_py-4.6.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.6 -prefs=runtime.tools.mkspiffs.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esp-rv32.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\esp-rv32\2302 -prefs=runtime.tools.esp-rv32-2302.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\esp-rv32\2302 -prefs=runtime.tools.esp-xs2.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\esp-xs2\2302 -prefs=runtime.tools.esp-xs2-2302.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\esp-xs2\2302 -prefs=runtime.tools.mklittlefs.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.xtensa-esp-elf-gdb.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\12.1_20231023 -prefs=runtime.tools.xtensa-esp-elf-gdb-12.1_20231023.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\12.1_20231023 -prefs=runtime.tools.riscv32-esp-elf-gdb.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gdb\12.1_20231023 -prefs=runtime.tools.riscv32-esp-elf-gdb-12.1_20231023.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gdb\12.1_20231023 -prefs=runtime.tools.esp32-arduino-libs.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67 -prefs=runtime.tools.esp32-arduino-libs-idf-release_v5.1-dc859c1e67.path=C:\Users\giaco\AppData\Local\Arduino15\packages\esp32\tools\esp32-ardu [... cut ...]
  114 |             (*((volatile uint32_t *)ETS_UNCACHED_ADDR(addr))) = (uint32_t)(val);                                       \

      |                                                                            ^~~

D:\Downloads\ESP32CompositeVideo-master\ESP32CompositeVideo-master\CompositeVideoSimple\CompositeOutput.h:179:5: note: in expansion of macro 'SET_PERI_REG_BITS'

  179 |     SET_PERI_REG_BITS(I2S_SAMPLE_RATE_CONF_REG(0), I2S_TX_BCK_DIV_NUM_V, 2, I2S_TX_BCK_DIV_NUM_S);

      |     ^~~~~~~~~~~~~~~~~

exit status 1

'I2S_CLKM_CONF_REG' was not declared in this scope

`

kestiq, 2024-09-15:
Had same issue.
Downgrading esp32 in boards manager to 2.0.0 worked for me.

MrC4ll, 2024-11-09:
Arduino IDE boards ESP32 ver 2.0.0 use depreciated declarations that may cause issues although it might still compile. Best course of action would be to read through and modify the declaration of I2S_CLKM_CONF_REG in newer versions.

sonikka666, 2025-04-14:
is there any update in this case ? i have the same problem.
i try to downgrade the esp32 ver 2.0.0,
if there is no sketch update, im afraid it still happen in the newest version of board or even the ide (cmiiw) 
some people (like me) is a newbie starting to use arduino/esp32 so i think the updated sketch with some adjustment (maybe) should be made.